ahtn/keyplus #15: uniflash.py big_endian_u16 is not defined

jschloer, 2018-01-09:
Using Python 3.5.2 and trying to run the uniflash info command, I get the failure at line 206. I can comment it out and keep moving, but I think I need the results of that to know whether to use the 16 or 32 hex file.

jschloer, 2018-01-09:
I now see that one of the more recent commits is labeled broken. Sorry for the noise. I'll keep an eye out for updates.

ahtn, 2018-01-11:
Should be fixed now.

Just a heads up, the nRF24LU1+ v0.2.0 build seems to drop a lot of packets when run on a Unifiying receiver. However, It works fine when I run it on my nRF24LU1+ dev board. So I still need to track down the cause of this bug before the it'll run nicely on a Unifying receiver.

closed 2018-01-11 by ahtn

ahtn, 2018-01-12:
> Just a heads up, the nRF24LU1+ v0.2.0 build seems to drop a lot of packets when run on a Unifiying receiver. However, It works fine when I run it on my nRF24LU1+ dev board. So I still need to track down the cause of this bug before the it'll run nicely on a Unifying receiver.

Turns out it was just a bad Unifying receiver, sorry for the confusion. If you try and get this running, it'd be great if you could tell me how it goes.

jschloer, 2018-01-12:
This looks to be working now. 
results:
```
flash_start?: 0x000
flash_end: 0x67ff
block_size: 0x0200
mcu: nRF24LU1+ rev. A
```