ahtn/keyplus #20: Abstract as much of the USB code as possible

ahtn, 2018-01-22:
Handling USB requests is handled differently on each platform. Need to implement a framework so that they can all share the same implementation.

eltang, 2018-04-15:
I think the best solution in the long term would be to use a prewritten USB library. I'm a big fan of ChibiOS's USB library, not least because it's the only full-featured interrupt-based USB driver for megaAVR I've found. Unfortunately, it doesn't support XMEGA or nRF24 at the moment, but it seems pretty straightforward to port it to a new architecture.

ahtn, 2018-04-15:
One issue with ChibiOS is it is GPL licensed.  I want to support Bluetooth in the future with Nordic's nRF52 microcontrollers. However, a lot of Nordic's code is released under licenses that are incompatible with the GPL, which would prevent me from redistributing it legally without obtaining a license from ChibiOS.

I've already done a bit of work on this. Most of the the custom endpoint code will work just by implementing [this interface](https://github.com/ahtn/keyplus/blob/master/src/usb_reports/usb_reports.h). The [USB descriptors](https://github.com/ahtn/keyplus/tree/master/src/usb) are shared too. 

A lot of the stuff that isn't shared between the different platforms now is localized in one file for each port. It's mostly just setup and handling the USB setup/control endpoint.

I'm not sure how far to take this.  One advantage of less abstraction is smaller code size, which will help on save space on 8-bit systems. The only thing I feel that really needs to be abstracted now is the HID interface requests and string descriptor requests. But it's not a major deal since, I don't expect them to change too frequently.

In the future for ARM, it might be worth looking into ARM MBED, I think it has an MIT compatible license.

eltang, 2018-04-16:
The ChibiOS HAL can be used standalone, separately from the main RTOS. The HAL is licensed under the Apache 2.0 license and is free for all purposes.

ahtn, 2018-04-23:
Cool that makes it useable then. It might be more work than it's worth to port it to some of the currently supported platforms, but it seems like it has very good support for the STM32 microcontrollers.  I have some STM32 dev boards lying around, so I might give it a go some time.

eltang, 2018-04-23:
If you think you can use the HAL to speed along AVR support then please do. I'd love to take keyplus for a spin on my ErgoDox EZ.