ahtn/keyplus #24: Figure out what is required to flash Unifying receivers from Windows

ahtn, 2018-01-22:
Unlike Linux/OS X, on Windows it is always necessary to have a driver to communicate with a USB device, so the `uniflash.py` script probably won't work on Windows without a driver for the unifying receiver.

To get the script to work, it might be as simple as installing the Logitech drivers for the Unifying receiver. If that doesn't work, it might be necessary to use [zadig](http://zadig.akeo.ie/) to use a generic libusb driver to communicate with the Unifying receiver.

lightingghost, 2018-05-23:
Hi,

I found out that the logitech driver does not work on windows. And I also tried zadig driver of libusb-win32 and libusbK. Both does not work. Is there any other setting that you recommend? 

Btw, do you have an instruction of communication with USB device on mac?

Thanks

angustrau, 2018-05-23:
Hi,
Flashing on Windows is still broken. You will have to try flashing from a Linux computer or VM.

lightingghost, 2018-05-24:
Thanks @angustrau . I am trying to flash from Ubuntu 18.04 VM, I am using VMware and connected the usb device to the VM. But it still cannot find the device.