ahtn/keyplus #40: Uniflash Issues

Merlin04, 2018-03-21:
``` $ sudo python3 uniflash.py info
[sudo] password for : 
Traceback (most recent call last):
  File "uniflash.py", line 372, in <module>
    detach_kernel_drivers(boot_dev, [0, 1])
  File "uniflash.py", line 33, in detach_kernel_drivers
    if dev.is_kernel_driver_active(i):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```

angustrau, 2018-03-22:
Make sure you have `libusb` and `hidapi` installed.

ahtn, 2018-03-22:
Some unifying receivers don't use the nRF24LU1+ which is needed for keyplus (I think they use the TI CC2544). I updated the script to better detect the bootloader (eded18e3a569eb67bed962a96082bc34a0cf3142).  You should be able to try again and it should tell you what microcontroller your receiver is using.

As far as I know, the low end unifying devices are more likely to come with an nRF24LU1+ unifying receiver.

ahtn, 2018-03-24:
For reference today I went and bought the cheapest Unifying mouse I could find (M235), and it's using the TI CC2544. I've bought the same model in the past before, and it was using an nRF24LU1+.

Merlin04, 2018-04-22:
I've figured out why the info command fails, it's because it can't find any devices, but then proceeds with the rest of the program:
```python3
        if (len(ti_devices) + len(nrf_devices) == 0):
            print("Couldn't find any devices to program", file=sys.stderr)
        elif (len(ti_devices) + len(nrf_devices) > 1):
            # another error
        elif (len(ti_devices) == 1):
            # unsupported message
            boot_dev = boot_devices[0]
            is_ti_device = True
        elif (len(nrf_devices) == 1):
            # found supported device
            boot_dev = boot_devices[0]

        detach_kernel_drivers(boot_dev, [0, 1])
```
So, it seems like the problem is that the unifying dongle isn't detected. I was able to confirm this by debugging this part. 
My `boot_devices` value is `[<DEVICE ID 046d:c52b on Bus 002 Address 028>]`. 

However, I've found another problem. When I use the "reset" command with uniflash, I get this:
```python3
Traceback (most recent call last):
  File "uniflash.py", line 376, in <module>
    detach_kernel_drivers(boot_dev, [0, 1])
  File "uniflash.py", line 50, in detach_kernel_drivers
    if dev.is_kernel_driver_active(i):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```
This is also failing because boot_dev is being passed into the is_kernel_driver_active function, and boot_dev is NoneType. It is NoneType because the unifying dongle isn't being detected, so the object `boot_dev = usb.core.find(idVendor=VID_LOGITECH, idProduct=pid)` returns is empty. 

It seems like the only actual problem is that the dongle isn't being detected.

Merlin04, 2018-04-22:
I figured it out, the issue was simple. 
I wasn't using the sleep in the example code. What worked for me was this:
```fish
sudo python3 uniflash.py icp; and sleep 2; and sudo python3 uniflash.py info
```
and then this:
```fish
sudo python3 uniflash.py icp; and sleep 2; and sudo python3 uniflash.py flash keyplus_nrf24lu1_unirecv-v0.3.0.hex
```

ahtn, 2018-04-22:
So did you need to change any of the code for this to work in the end?

Merlin04, 2018-04-22:
No, I didn't need to. I'll probably modify the code to make the error more clear.

Merlin04, 2018-05-03:
It seems that if I replace the `uniflash.py icp` with `keyplus-cli bootloader`, it solves the `is_kernel_driver_active` issue, but when I try to flash the receiver, I get this:
```python3
 > sudo python3 uniflash.py flash keyplus_nrf24lu1-unirecv.hex 
Found a supported device with bootloader: (0x46d, 0xaaaa)
flash_start?: 0x0000
flash_end: 0x67FF
block_size: 0x0200
mcu: b'nRF24LU1+ rev. A'
Packet0:
00000000: 60 00 00 02 2A DF 46 32  34 4C 55 31 2B 20 72 65  `...*.F24LU1+ re
00000010: 76 2E 20 41 00 00 00 00  00 00 03 00 15 19 00 00  v. A............
27677 26622
Traceback (most recent call last):
  File "uniflash.py", line 462, in <module>
    boot_writer.write_hexfile(fileName)
  File "uniflash.py", line 331, in write_hexfile
    raise HexfileTooBig("hexfile too big")
__main__.HexfileTooBig: hexfile too big
```
I've tried all of the hex files from ci.keyplus.io, but none work.

ahtn, 2018-05-03:
To flash hex files to the unifying receiver, they need to have no layout/settings pre loaded (ones on ci.keyplus.io don't atm).

I'll try update the release builds soon, but here's a hex file that should work:

[keyplus_nrf24lu1-unirecv-blank_v0.3.1_pre-00d9cc0.hex.zip](https://github.com/ahtn/keyplus/files/1970604/keyplus_nrf24lu1-unirecv-blank_v0.3.1_pre-00d9cc0.hex.zip)

Merlin04, 2018-05-03:
Thanks, that works.

closed 2018-05-03 by Merlin04

lightingghost, 2018-05-25:
I got exactly the same error here:

```python
sudo python3 uniflash.py info
Found a supported device with bootloader: (0x46d, 0xaaaa)
Traceback (most recent call last):
  File "uniflash.py", line 446, in <module>
    boot_info = boot_writer.cmd_bootloader_info()
  File "uniflash.py", line 221, in cmd_bootloader_info
    response = self.simple_cmd(CMD_BOOTLOADER_INFO)
  File "uniflash.py", line 217, in simple_cmd
    self.send_packet(cmd_packet)
  File "uniflash.py", line 204, in send_packet
    set_report_request(self.dev, 0, 0, packet)
  File "uniflash.py", line 88, in set_report_request
    return dev.ctrl_transfer(bmRequestType, SET_REPORT, wValue, interface, data)
  File "/home/odin/.local/lib/python3.6/site-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/home/odin/.local/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/odin/.local/lib/python3.6/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/home/odin/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/home/odin/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```

I did not forget to `sleep 2`. Can I ask how you solve this error? @Merlin04