ahtn/keyplus #45: Document how to make rf_settings part of layout file

Merlin04, 2018-04-14:
I'm really confused about how to do this, at the top of the example file it says "Generated on `date` on `time`", is there some program I run to generate this?

ahtn, 2018-04-14:
You can use the keyplus flasher to generate this file. The latest version is [on the releases page](https://github.com/ahtn/keyplus/releases). There's only a build for windows at the moment, but you can also run the flasher on Linux by cloning the git repository and running form source.

Choose  "Device and RF settings" from the drop down box and then you should be able to generate and program the RF settings to your devices.

[image]

The reason for keeping this file separate is it contains your encryption key for your devices, so you'll want to generate your own and keep it private. After you have loaded your RF settings file, you only need to flash your layout file to the device as the device will preserve the previously loaded settings.

Merlin04, 2018-04-14:
Since keyplus flasher isn't working on my computer, is there a way to generate it from the command line or manually?