ahtn/keyplus #47: make program-boot fails

Merlin04, 2018-04-14:
```
 > sudo make BOARD=keyplus_mini program-boot
python3 ../host-software/keyplus-cli program \
	 \
	--new-id 0 \
	--layout "../layouts/basic_split_test.yaml" \
	--rf "../layouts/test_rf_config.yaml" \
	--fw-hex "build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u.hex" \
	-M 0x7600 0x7800 0x0800 \
	-o ""build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-basic_split_test.hex"" \
	-F chip_name="ATxmega32A4U" \
	-F scan_method=fast_row_col \

# ./scripts/flash_hex.sh "build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-basic_split_test.hex" atxmega32a4u
python3 ../host-software/keyplus-cli bootloader -p ATxmega32A4U | sleep 1.5
Error: Couldn't find any matching devices.
python3 ./xusb-boot/scripts/bin/xusbboot-cli -f "build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-basic_split_test.hex" -mcu atxmega32a4u
Traceback (most recent call last):
  File "./xusb-boot/scripts/bin/xusbboot-cli", line 183, in <module>
    if (dev.vendor_id, dev.product_id) in xusbboot.DEFAULT_USB_IDS:
AttributeError: module 'xusbboot' has no attribute 'DEFAULT_USB_IDS'
Makefile:227: recipe for target 'program-boot' failed
make: *** [program-boot] Error 1
```

Merlin04, 2018-04-22:
The xusb issue still persists even after following your advice.

ahtn, 2018-04-22:
Looks like I forgot to update that package. I pushed the new version and you should be able to update it now with:

```
sudo pip install -U xusbboot
```

closed 2018-05-03 by ahtn