ahtn/keyplus #48: v0.2.2 doesn't detect any devices; can't use 0.3

Merlin04, 2018-04-14:
I can't use version 0.3 because my keyplus mini boards have version 0.2.2, and I don't have an avr programmer to update it. The keyplus app doesn't detect any devices. No error messages are shown.

Merlin04, 2018-04-14:
For some reason, the 0.3 programmer works with my board running 0.2.2, is this supposed to happen?

angustrau, 2018-04-14:
You can update your 0.2.2 board to 0.3 from the firmware update section of keyplus flasher. The 0.3 flasher should work fine with 0.2 boards. Keyplus must have not released the USB device properly last time you closed it leading to your last error. Restarting your computer should fix it.

Merlin04, 2018-04-14:
Can I update it if I don't have an AVR programmer?

angustrau, 2018-04-14:
Yes, just go to the firmware update tab, select your hex, and press program.

Merlin04, 2018-04-14:
When I try to, I get this error:
[image]
Nothing is printed in the terminal.

ahtn, 2018-04-14:
Try press program again and it should work.  I think the device just took longer too enter its bootloader than expected.

If you are running from source, it will be a good idea to checkout the specific git commit for the version you are using. You can use `git checkout v0.3.0`, to checkout a specific version.

Like was mentioned before you should be able to use v0.3.0 to update the firmware hex.  Note on the rev3 controllers, I was meaning to bump the firmware version to v0.3.0 before I sent them out, but I forgot.  So they say they are v0.2.2 but should actually be v0.3.0. I'll make sure to add a note about this somewhere.

Merlin04, 2018-04-22:
That worked, I forgot to close the issue.

closed 2018-04-22 by Merlin04