ahtn/keyplus #61: Right half of split keyboard not working

Merlin04, 2018-04-29:
For some reason the right half of my keyboard isn't working when plugged in via usb or when wireless. I tried making it the left half, which seemed to work via usb, but didn't work using wireless. I'm not sure what's happening, here's my layout file:
```yaml
# ss2-pcb-unified

report_mode: auto_nkro

devices:
  "SuperSplit - Left":
    id: 0
    layout: "SuperSplit"
    layout_offset: 0
    wireless_split: true
    wireless_mouse: true
    wired_split: false
    scan_mode:
      mode: col_row
      rows: 6
      cols: 6
      # maps how keys are physically wired, to how they appear visually
      matrix_map: [
        r0c0, r0c1, r0c2, r0c3, r0c4, r0c5,
        r1c0, r1c1, r1c2, r1c3, r1c4, r1c5,
        r2c0, r2c1, r2c2, r2c3, r2c4, r2c5,
        r3c0, r3c1, r3c2, r3c3, r3c4, r3c5,
        r4c0, r4c1, r4c2, r4c3, r4c4, r4c5,
        r5c0, r5c1, r5c2, r5c3, none, r5c5
      ]
  
  "SuperSplit - Right":
    id: 1
    layout: "SuperSplit"
    layout_offset: 1
    wireless_split: true
    wireless_mouse: true
    wired_split: false
    scan_mode:
      mode: col_row
      rows: 6
      cols: 9
      # maps how keys are physically wired, to how they appear visually
      matrix_map: [
        r0c0, r0c1, r0c2, r0c3, r0c4, r0c5, r0c6, r0c7, r0c8,
        r1c0, r1c1, r1c2, r1c3, r1c4, r1c5, r1c6, none, r1c8,
        r2c0, r2c1, r2c2, r2c3, r2c4, r2c5, r2c6, r2c7, r2c8,
        r3c0, r3c1, r3c2, r3c3, r3c4, r3c5, r3c6, r3c7, r3c8,
        r4c0, r4c1, r4c2, r4c3, r4c4, r4c5, r4c6, r4c7, r4c8,
        r5c0, none, r5c2, r5c3, r5c4, r5c5, r5c6, r5c7, r5c8
      ]
  nrf_dongle:
    id: 16
    wireless_split: true
    wireless_mouse: true
    scan_mode: # rf dongle, so don't need a matrix
      mode: no_matrix # options: col_row, pins, no_matrix

layouts:
  supersplit:
    default_layer: 0
    layers: [
      [ # Layer 0
        [ # Left Half
          esc, f1, f2, f3, f4, f5,
          grave, 1, 2, 3, 4, 5,
          tab, q, w, e, r, t,
          caps, a, s, d, f, g,
          lsft, z, x, c, v, b,
          lctl, lgui, lalt, bspc, none, SPC
        ],
        [ # Right Half
          f6, f7, f8, f9, f10, f11, f12, home, end,
          6, 7, 8, 9, 0, '-', '=', none, bspc,
          y, u, i, o, p, '[', ']', '\', del,
          h, j, k, l, ';', "'", ent, pgup, home,
          n, m, ',', '.', '/', rsft, up, pgdn, end,
          SPC, none, L1, menu, rctl, left, down, right, ins
        ]
      ],
      [ # Layer 1
        [ # Left Half
          esc, f1, f2, f3, f4, f5,
          grave, 1, 2, 3, 4, 5,
          tab, q, w, e, r, t,
          caps, a, s, d, f, g,
          lsft, z, x, c, v, b,
          lctl, lgui, lalt, bspc, none, SPC
        ],
        [ # Right Half
          f6, f7, f8, f9, f10, f11, pscr, slck, pause,
          6, 7, 8, 9, 0, '-', '=', none, bspc,
          y, u, i, o, p, '[', ']', '\', del,
          h, j, k, l, ';', "'", ent, pgup, home,
          n, m, ',', '.', '/', rsft, up, pgdn, end,
          SPC, none, L1, menu, rctl, left, down, right, ins
        ]
      ]
    ]
```

ahtn, 2018-04-30:
I found a bug recently, that causes RF settings not to be updated until the device is reset. It's possible it might be causing this issue.

Not at home at the moment, so don't have all the stuff I need to test this, so might not get a chance to test this for the next day or two.

ahtn, 2018-05-01:
Found a bug in the calculation of the matrix offset for split devices. Since the resulting offset was too large attempts to write to the matrix where blocked as they were out of bounds.

Fixed with this commit: https://github.com/ahtn/keyplus/commit/9d658e2dbcadba9b947eb310d67d098732b42ff4

Merlin04, 2018-05-03:
I've tried compiling the firmware for the unifying dongle, but I just get errors saying `/bin/bash: /usr/local/bin/sdcpp: No such file or directory`.

Merlin04, 2018-05-03:
Never mind, I didn't realize ci.keyplus.io existed.

closed 2018-05-03 by Merlin04