ahtn/keyplus #67: Compilation errors with nrf24lu1

Merlin04, 2018-05-03:
```fish
 > make BOARD=unirecv
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_normal.h:6,
                 from ../src/key_handlers/key_normal.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_mouse.h:6,
                 from ../src/key_handlers/key_mouse.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_mouse.c:6:
/usr/include/string.h:36:42: error: missing binary operator before token "("
/usr/include/string.h:164:21: error: missing binary operator before token "("
/usr/include/string.h:173:43: error: missing binary operator before token "("
/usr/include/string.h:491:19: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_media.h:6,
                 from ../src/key_handlers/key_media.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_macro.h:6,
                 from ../src/key_handlers/key_macro.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_hold.h:6,
                 from ../src/key_handlers/key_hold.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_hold.c:6:
/usr/include/string.h:36:42: error: missing binary operator before token "("
/usr/include/string.h:164:21: error: missing binary operator before token "("
/usr/include/string.h:173:43: error: missing binary operator before token "("
/usr/include/string.h:491:19: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_handlers.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/key_handlers/key_handlers.h:12,
                 from ../src/key_handlers/key_custom.h:6,
                 from ../src/key_handlers/key_custom.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/usb/util/requests.h:6,
                 from ../src/usb/device_descriptors.c:6:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/core/util.h:11,
                 from ../src/usb_reports/mouse_report.h:6,
                 from ../src/usb_reports/mouse_report.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/usb_reports/mouse_report.c:9:
/usr/include/string.h:36:42: error: missing binary operator before token "("
/usr/include/string.h:164:21: error: missing binary operator before token "("
/usr/include/string.h:173:43: error: missing binary operator before token "("
/usr/include/string.h:491:19: error: missing binary operator before token "("
In file included from ../src/core/util.h:11,
                 from ../src/usb/descriptors.h:6,
                 from ../src/usb_reports/vendor_report.h:6,
                 from ../src/usb_reports/vendor_report.c:4:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from ../src/core/debug.h:14,
                 from ../src/core/ring_buf.h:8,
                 from ../src/usb_reports/vendor_report.h:9,
                 from ../src/usb_reports/vendor_report.c:4:
/usr/include/stdlib.h:131:36: error: missing binary operator before token "("
/usr/include/stdlib.h:174:17: error: missing binary operator before token "("
/usr/include/stdlib.h:188:36: error: missing binary operator before token "("
In file included from ../src/usb_reports/vendor_report.c:126:
/usr/include/string.h:36:42: error: missing binary operator before token "("
/usr/include/string.h:164:21: error: missing binary operator before token "("
/usr/include/string.h:173:43: error: missing binary operator before token "("
/usr/include/string.h:491:19: error: missing binary operator before token "("
In file included from ../src/usb_reports/media_report.h:6,
                 from ../src/usb_reports/media_report.c:4:
[... 207 lines cut ...]
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
In file included from aes.c:6:
/usr/include/string.h:36:42: error: missing binary operator before token "("
/usr/include/string.h:164:21: error: missing binary operator before token "("
/usr/include/string.h:173:43: error: missing binary operator before token "("
/usr/include/string.h:491:19: error: missing binary operator before token "("
make: *** No rule to make target 'build/unirecv/obj/aes.dep', needed by 'build/unirecv/keyplus_nrf24lu1-unirecv.hex'.  Stop.
```
When I run it again:
```fish
compiling: aes.c
compiling: flash.c
compiling: hardware.c
compiling: led.c
compiling: main.c
compiling: nonce.c
nonce.c:50: error 78: incompatible types
from type 'unsigned-int near* fixed'
  to type 'unsigned-char generic* fixed'
nonce.c:51: error 78: incompatible types
from type 'const-unsigned-long-int near* fixed'
  to type 'unsigned-char generic* fixed'
Makefile:206: recipe for target 'build/unirecv/obj/nonce.rel' failed
make: *** [build/unirecv/obj/nonce.rel] Error 1
```

ahtn, 2018-05-03:
The efm8 branch has [scripts](https://github.com/ahtn/keyplus/blob/00d9cc0f0b0f97c4b32ba94a522d521041ea08cf/env/ubuntu-14.04-setup.sh#L24) for building on nrf24lu1+ (it uses the SDCC compiler too). When it's ready I'll merge it to master and documentation on how to build it.

With the unifying receivers, need to be a little bit careful since if you load a bad build it is quite hard to recover it again.

denwi, 2021-01-20:
I'm guessing this never got resolved? Can't find the efm8 branch now, and still getting the same errors on master :(