ahtn/keyplus #71: Flashing firmware 0.3.2 stuck at empty settings

BillyBuerger, 2018-05-16:
I flashed the latest firmware 0.3.2 on my Keyplus Rev03 and it appears to flash fine but the info states "Warning: Empty settings!"  Every time I've flashed the firmware I end up with a device ID set to 255.  So I go to Device and RF, pick a layout file and set Device id to 0 and generate a new RF settings, then program.  This time I get an error "'encoding' is an invalid keyword argument for this function".  I am able to flash back to firmware 0.3.1 without any issues and get it back working again.

[image]

angustrau, 2018-05-16:
The device settings resetting every time you flash is an unfortunate side effect of keyplus storing its settings in the flash section of the xmega instead of eeprom.

The `'encoding' is an invalid keyword argument for this function` error is likely because Python 2 (which the Windows version of the flasher is built with) version of `open()` [doesn't support the `encoding` argument](https://docs.python.org/2/library/functions.html#open) introduced [in this commit](https://github.com/ahtn/keyplus/commit/dff4ac9cd8de4384829830e38b7012f68e4cc3c1#diff-14ec0fd69cc43e190c92bcf013139160L55).

angustrau, 2018-05-16:
Try using [this build](https://ci.keyplus.io/job/keyplus-flasher-angustrau/lastSuccessfulBuild/artifact/keyplus/host-software/dist/keyplus_flasher-0.3.2-c084625-build5.exe).

angustrau, 2018-05-16:
Hmm. You should be able to download it from [here](https://ci.keyplus.io/job/keyplus-flasher/)

angustrau, 2018-05-16:
This appears to be a bug with Keyplus Flasher's port to PyQt5. It seems to be fixed in the latest build from Keyplus CI.

BillyBuerger, 2018-05-17:
Thanks, that did it.  I was able to flash the 3.3 firmware without issue.  And now my volume keys are working.  Thanks for the hard work.

PR #72 (2018-05-16): Fix python 2 support in layout parser
Fixes #71
closed 2018-05-17 by BillyBuerger