ahtn/keyplus #83: How to debug wireless connection

lightingghost, 2018-05-26:
Firstly I would like to thank you for making this great project. But I came across a problem when I am trying to build a wireless split keyboard. The wired mode works fine, but the wireless mode does not work. 

What I have done:

1. Flash the logitech unifying receiver using the follow command:
```bash
sudo python3 uniflash.py icp
sudo python3 uniflash.py flash keyplus_nrf24lu1-unirecv-0.3.3.hex
```
2. Flash the keyplus mini controller using `keyplus_flasher.py`.
3. Generate RF settings and load layout under the `device and RF` tab.
Now the keyplus_flasher gui show the following result:

[image]

Now the wired mode is working, but the wireless mode is not. 
Here is my configuration:
```yaml
name: "test-layout"
version: 0.3.0
report_mode: auto_nkro # options: auto_nkro, 6kro, nkro

devices:

  test0:
    id: 0
    wireless_split: true
    wireless_mouse: true
    wired_split: false
    layout: test_layout
    layout_offset: 0
    scan_mode:
      mode: col_row # options: col_row, pins, none
      rows: 5
      cols: 7
      # maps how keys are physically wired, to how they appear visually
      matrix_map: [
        r0c0, r0c1, r0c2, r0c3, r0c4, r0c5, r0c6, 
        r1c0, r1c1, r1c2, r1c3, r1c4, r1c5, r1c6,
        r2c0, r2c1, r2c2, r2c3, r2c4, r2c5, r2c6,
        r3c0, r3c1, r3c2, r3c3, r3c4, r3c5, r3c6,
        r4c0, r4c1, r4c2, r4c3, r4c4, r4c5, r4c6,
      ]

  # rf dongle
  nRF24 Dongle:
    id: 63
    wireless_split: true
    wireless_mouse: true
    scan_mode: # rf dongle, so don't need a matrix
      mode: no_matrix # options: col_row, pins, no_matrix

layouts:
  test_layout:
    layers: [
      [ # layer 0 (base)
        [
          a, b, c, d, e, f, 1,
          g, h, i, j, k, l, 2,
          m, n, o, p, q, r, 3,
          s, t, u, v, w, x, 4,
          y, z, 5, 6, 7, 8, 9,
        ]
      ],
]
```
Is there a way to debug whether the keyboard is sending wireless packets, and whether the dongle is receiving packets?

Any help will be appreciated.

Merlin04, 2018-05-28:
`version: 0.3.0`
I don't know if this is the problem, but try changing this to 0.3.3.

lightingghost, 2018-05-29:
Turns out that 2.95V is not a good battery. T^T Thanks for helping @Merlin04

closed 2018-05-29 by lightingghost