ahtn/keyplus #84: wireless connection quality

lightingghost, 2018-06-12:
Is there a way to check the wireless connection quality?

The left part of my keyboard sometimes send multiple repeated keys. I assume it is related to the wireless connection between the left half and the receiver. Is there a way to improve the wireless connection quality?

Thanks

angustrau, 2018-06-12:
There isn't a way to debug the wireless connection, but there's a good chance your issue isn't related to the wireless. You may be experiencing chattering, which is especially prevalent on box switches. Try adjusting your debounce settings to compensate.

lightingghost, 2018-06-12:
@angustrau Thanks! I did use box switches for my setup. I will have a try on the debounce settings.

Algram, 2020-02-22:
@lightingghost I have the same issue and I don't think it has to do with debouncing. Did you manage to solve it?