ahtn/keyplus #86: Compile for 16kB dongles

bezmi, 2018-06-26:
Hi, I've got one of those cheap aliexpress 16kB nrf24lu1 dongles and I'd like to know if there is some way to compile a minimal working example of the firmware (hardcoded keymap, rf settings, etc) so I can use the dongle for development.

I've just built a nrf51822 split wireless keyboard system and would love to integrate it with your firmware so I can use a smaller dongle, but I don't have time to wait for a CU-0007 logitech receiver to travel from China to QLD, Australia before the university break is over... Otherwise, do you know of any cheap-ish logitech mice that are currently sold which are guaranteed to utilise the nrf24lu1 receiver?

angustrau, 2018-06-26:
Keyplus currently doesn't support 16kB dongles. The M235 appears to have the nrf24 receiver and it's on sale at [JBHiFi](https://www.jbhifi.com.au/computers-tablets/accessories/logitech/logitech-m235-wireless-mouse-blue/91050/)

bezmi, 2018-06-27:
Awesome, I'll go pick one up today!

bezmi, 2018-06-27:
So I went and picked up an M235 today and it's a  CU-0008 receiver with firmware version 024.001.00023 :(

angustrau, 2018-06-27:
That's too bad :(
I have a spare receiver I can let go for the cost price, I'm in Melbourne. Give me a pm on reddit (/u/amtra5) if you're interested.