ahtn/keyplus #87: Is Keyplus dead?

Merlin04, 2018-09-26:
I haven't seen any activity on this for a while.

serverside-is, 2018-09-26:
Been asking myself the same. Wondering what happened to the repository owner. "Disappeared" so suddenly. Hope he is doing fine.

jockus, 2018-10-09:
Just wanted to chime in I've been eagerly awaiting further drops of the keyplus chipset, and would help in any way possible to keep this going.

LovesTha, 2018-10-11:
First thing I'd like to design and build is a numpad PCB, do people think that would be desired by many?

gawlk, 2018-10-11:
If someone could manage to add the hold/tap feature that would be awesome ! My only mk (a gherkin) is basically taking the dust because it is unfortunately unusable without it.

jockus, 2018-10-12:
@LovesTha I think the most useful pcb to get going to begin with is one designed for handwiring, same as the keyplus pcbs that _spindle was selling on [mechmarket](https://www.reddit.com/r/mechmarket/comments/86zhpa/au_h_keyplus_usb_typec_keyboard_controllers_w/?utm_content=title&utm_medium=hot&utm_source=reddit&utm_name=friends).

Since the codebase targets the ATXMEGA 32AU4 chipset, which seems to not really have any readily available PCBs to purchase, just getting a small pcb that works with keyplus is step one.

It would be interesting to see if the schematics that were used for the keyplus pcbs are available somewhere.

angustrau, 2018-10-12:
The PCB layout files and schematics for the keyplus mini are available [here](https://github.com/ahtn/keyboard_pcb/tree/master/keyplus_mini/keyplus_mini_v3).

jockus, 2018-10-12:
So it is! Time to look in to how to get some PCBs manufactured.

LovesTha, 2018-10-12:
Jockus, I don't disagree with your reasoning but it isn't a board I'm excited to own myself.

georgearnall, 2019-01-15:
I'd be happy to lend a hand too, I was hoping to get my hands on one of the keyplus mini's but missed the order. Was potentially looking at fabricating my own.

Does anyone know what happened to _spindle?

LovesTha, 2019-01-18:
No idea what happened to spindle, just assuming he got busy.

I'm nearly ready to order my version of the AlphaSplit, but with a few more layout options:
* Keycap compatible bottom row (125/125/125/125/100 and 100/125/125/125/125)
* Numpad layout, both landscape and portait. Normal 4x4 numpad layout with either 2 extra rows on the top or left. If you do the extra rows on top then all keys are rotated 90degrees and the USB ports are on the left.

LovesTha, 2019-01-18:
Oh and maybe with rotary encoders in the bottom corners.....

georgearnall, 2019-01-18:
If you put together a layout and get it fabricated , I would definitely be interested

LovesTha, 2019-01-18:
https://imgur.com/a/b8hNp8x for an idea of where I'm up to, or check out my fork of the PCB project.
Although a sensible MOQ barely deals with my desired boards. But I'm willing to do a second run for others as a kit (or hand soldered for substantially more cash). Getting boards machine soldered is a jump I'm not sure I want to do.

ahtn, 2019-04-01:
Hey, I'm back.  Sorry for disappearing so suddenly. :( I hope to catch up with things over the coming days.

serverside-is, 2019-04-01:
We don't know each other, but believe me that this notice made my day. I was truly convinced that something very bad and definitive had happened to you, and thought about it sometimes.
Welcome back!! :-D

LovesTha, 2019-04-01:
Happy to see you back, hopefully I can get some of my LTOmni boards soldered up soon

h3lgi, 2019-06-26:
Happy to see you because I'm interested to buy KeyPlus :)