ahtn/keyplus #93: Intehex Module

LovesTha, 2019-04-14:
Attempting to compile I get an error about the intelhex module not being found:

### Build settings ###
Compiling C: ../../src/core/crc.c
Compiling C: ../../src/core/error.c
TARGET:         xmega_keyplus-keyplus_mini-atxmega32a4u
Compiling C: ../../src/core/flash.c
Compiling C: ../../src/core/hardware.c
BOARD:          keyplus_mini
Compiling C: ../../src/core/layout.c
Compiling C: ../../src/core/packet.c
Compiling C: ../../src/core/ring_buf.c
MCU:            atxmega32a4u
Compiling C: ../../src/core/util.c
LAYOUT_FILE:    /home/gareth/keyboard/KeyPlus/keyplus/layouts/alpha_split.yaml
Compiling C: ../../src/core/settings.c
Compiling C: ../../src/core/nrf24.c
Compiling C: ../../src/core/rf.c
RF_FILE:        /home/gareth/keyboard/KeyPlus/keyplus/layouts/test_rf_config.yaml
Compiling C: ../../src/core/nonce.c
Compiling C: ../../src/core/io_map.c
Compiling C: ../../src/core/unifying.c
compiler c flags:
no-error=unused-but-set-variable -fno-jump-tables -g -DVBUS_PIN_PORT=E -DVBUS_PIN_NUM=2 -DVBUS_PIN_INT_NUM=1 -DNRF24_CE_PORT=PORTR -DNRF24_CE_PIN=PIN0_bm -DNRF24_IRQ_PIN_PORT=R -DNRF24_IRQ_PIN_NUM=1 -DNRF24_IRQ_INT_NUM=1 -DMCU_FLASH_SIZE=32 -DMCU_STRING="ATxmega32A4U" -DXMEGA_PIN_COUNT=44 -DBUILD_TIME_STAMP=0x1ER_ADDR= -DUSE_NRF24=1 -DNONCE_ADDR= -DSUPPORT_GESTURE=1 -DUSE_UNIFYING=1 -DUSE_I2C=1 -DUSE_SCANNER=1 -DMAX_NUM_ROWS=18 -DUSE_HARDWARE_SPECIFIC_SCAN=0 -DSUPPORT_MACRO=1 -DUSE_USB=1 -DUSE_BLUETOOTH=0 -DHAS_MOUCompiling C: ../../src/core/usb_commands.c
Compiling C: ../../src/core/keycode.c
Compiling C: ../../src/core/macro.c
Compiling C: ../../src/usb_reports/mouse_report.c
Compiling C: ../../src/usb_reports/vendor_report.c
Compiling C: ../../src/usb_reports/media_report.c
aliasing -Wstrict-prototypes -Wa,-adhlns=print_keyplus_info -I./src/ -std=gnu99  -Wall -Werror -MMD -MP -MF build/keyplus_mini-atxmega32a4u/obj/.dep/print_keyplus_info.d
Compiling C: ./src/xmega_hardware/twi_slave_driver.c
Compiling C: ./src/xmega_hardware/twi_master_driver.c
Compiling C: ./src/aes.c                                                                        [0/1783]
Compiling C: ./src/usb_reports.c                                                                        
../../src/core/rf.c:230:22: warning: ‘last_crc’ defined but not used [-Wunused-variable]                
 static XRAM uint16_t last_crc[NRF24_NUMBER_PIPES];                                                     
                      ^                                                                                 
=== ELF Size Information (xmega_keyplus-keyplus_mini-atxmega32a4u) ===                                  
../../src/usb_reports/vendor_report.c:152:21: warning: ‘first_sent’ defined but not used [-Wunused-variable]                          
 static XRAM uint8_t first_sent = false;  
                     ^               
   text    data     bss     dec     hex filename
  31013     206    3036   34255    85cf build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u.elf                                       
=== Size Information (xmega_keyplus-keyplus_mini-atxmega32a4u) ===
flash size: 29646                                   
flash delta: 29646                                  
ram size: 3242                                                                                          
ram delta: 3242                                                                                        
                                                                 
Linking: build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u.elf                    
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld:./ld-scripts/avrxmega2.xn:191: warning: memory region `key_settings_block' not declared                      
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld:./ld-scripts/avrxmega2.xn:195: warning: memory region `key_layout_block' not declared                       
Creating load file for Flash: build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u.hex                                                      
                                                     
### Merging layout settings into hex file ###  
python3 ../../src/../host-software/keyplus-cli program \
         \                             
        --new-id 0 \                   
        --layout "/home/gareth/keyboard/KeyPlus/keyplus/layouts/alpha_split.yaml" \
        --rf "/home/gareth/keyboard/KeyPlus/keyplus/layouts/test_rf_config.yaml" \
        --fw-hex "build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u.hex" \
        -M 0x7A00 0x7C00 0x0400 \                                             
        -o "build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-alpha_split.hex" \
        -F chip_name="ATxmega32A4U" \
        -F scan_method=fast_row_col \
        -F max_rows=18 \                                                         
                                    
Traceback (most recent call last):  
  File "../../src/../host-software/keyplus-cli", line 6, in <module>
    import intelhex                 
ModuleNotFoundError: No module named 'intelhex'
../../src/keyplus.mk:25: recipe for target 'build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-alpha_split.hex' failed                                                                     
make: *** [build/keyplus_mini-atxmega32a4u/xmega_keyplus-keyplus_mini-atxmega32a4u-alpha_split.hex] Error 1                                                   

Any clues? (It's confusing me a little as I'm not expecting that this compile is targeting intel hardware.

angustrau, 2019-04-14:
You need to install the python library "intelhex". `pip install intelhex`

LovesTha, 2019-04-15:
So having run that I still get the exact same error (did a couple of make cleans to be safe).

angustrau, 2019-04-16:
You might have multiple versions of python installed, so the dependency might not be installed for python 3. Try `python3 -m pip install intelhex --user`.

LovesTha, 2019-04-16:
Probably, it definitely was 2.7 that pip uninstalled it from when I uninstalled then reinstalled that library.

Now how to figure out how to install pip for python3 :)

> /usr/bin/python3: No module named pip

LovesTha, 2019-04-16:
PS: I suspect that is a problem I can solve, now I know I need to do things under python3 and not the system default. Will probably suggest some README.md patches to guide through this at some point :)

ahtn, 2019-06-04:
I updated the instructions to explicitly mention using `pip3`. I also updated all the scripts to explicitly call `#!/bin/env python3` instead of relying on the system default.  So I hope things are less confusing now.

closed 2019-06-04 by ahtn

LovesTha, 2019-06-04:
Sounds like ideal fixes.

If only I'd find the time and motivation to get one of my boards working. Hand soldering this stuff was a mistake.