ahtn/keyplus #101: EraseFailed when flash a unirecv, Error code: 49

kahoch, 2019-10-02:
Hi, 
When I trying to flash firmware into a logi unirecv, got error 49.
Any idea? Thank you!

[kaho@arch uniflash]$ sudo ./uniflash.py flash keyplus_nrf24lu1-unirecv-0.3.3.hex
Found a supported device with bootloader: (0x46d, 0xaaaa)
flash_start?: 0x0000
flash_end: 0x63FF
block_size: 0x0200
./uniflash.py:450: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
  mcu_str = boot_writer.cmd_get_mcu_str()
mcu: b'nRF24LU1+ rev. A'
Packet0:
00000000: 60 00 00 02 9C 00 46 32  34 4C 55 31 2B 20 72 65  `.....F24LU1+ re
00000010: 76 2E 20 41 3E 65 4C EF  40 4A 0B 8B 9E DD 5A 5F  v. A>eL.@J....Z_
23616 26622
Traceback (most recent call last):
  File "./uniflash.py", line 461, in <module>
    boot_writer.write_hexfile(fileName)
  File "./uniflash.py", line 338, in write_hexfile
    self.cmd_erase_page(i)
  File "./uniflash.py", line 305, in cmd_erase_page
    raise EraseFailed("Failed to erase page. Error code: {}".format(response[0]))
__main__.EraseFailed: Failed to erase page. Error code: 49

angustrau, 2019-10-02:
What does `uniflash info` give you?

kahoch, 2019-10-02:
[kaho@arch uniflash]$ sudo ./uniflash.py info
Found a supported device with bootloader: (0x46d, 0xaaaa)
flash_start?: 0x0000
flash_end: 0x63FF
block_size: 0x0200
./uniflash.py:450: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
  mcu_str = boot_writer.cmd_get_mcu_str()
mcu: b'nRF24LU1+ rev. A'

angustrau, 2019-10-02:
And you've run `uniflash icp` to get it into bootloader mode?

kahoch, 2019-10-02:
Yes, run uniflash icp and then uniflash info.

kahoch, 2019-10-02:
Hi, @angustrau 
I missed an error when I run `uniflash.py icp`, 😓

```
[kaho@arch uniflash]$ sudo ./uniflash.py icp
Traceback (most recent call last):
  File "./uniflash.py", line 366, in <module>
    enter_icp()
  File "./uniflash.py", line 107, in enter_icp
    detach_kernel_drivers(dev, [0, 1, 2])
  File "./uniflash.py", line 48, in detach_kernel_drivers
    if dev.is_kernel_driver_active(i):
AttributeError: 'NoneType' object has no attribute 'is_kernel_driver_active'
```
This is similar to Uniflash Issues #40, @Merlin04 said 

>  if I replace the uniflash.py icp with keyplus-cli bootloader, it solves the is_kernel_driver_active issue

 but if I use `keyplus-cli bootloader`, nothing happen, then I use `keyplus-cli list`, seems it can't find the unifying receiver.

kahoch, 2019-10-09:
Solved.
That receiver come with a signed bootloader. I bought a new one.
@angustrau Thank you for your help!

closed 2019-10-09 by kahoch