ahtn/keyplus #102: Layout example for directly wired keys without matrix

Algram, 2020-01-06:
Hi there,
first of all: This projects looks absolutely amazing. So good in fact that I started to design a pcb based on it for my gamepad.

In my version, every key is wired directly to a pin on the MCU. Here is a schematic:
[image]

Does keyplus currently support a config like this, and if it does, could you maybe provide an example for me?

Thanks a lot!

angustrau, 2020-01-06:
The project hasn't been active for a while now, but it does support your configuration. Have a look at the Spectre PCB.

[PCB](https://github.com/ahtn/keyboard_pcb/tree/master/spectre)
[Layout](https://github.com/ahtn/keyplus/blob/master/layouts/spectre.yaml)

Algram, 2020-01-06:
@angustrau very nice, so I could basically adapt the `layout.yaml` to my needs and use it in the keyplus firmware? I am using a different MCU than the specte but that does not seem to be a problem?

angustrau, 2020-01-06:
That's right. The 32a4u is a more or less equivalent MCU to the one in the spectre, and has pre-built firmware binaries available.

Algram, 2020-01-06:
@angustrau perfect, I am very curious how my pad will turn out, the PCB and parts are already ordered. I am especially hoping for the lowest latency possible and I hope this can be achieved with keyplus.