ahtn/keyplus #104: How to debug wireless issues?

Algram, 2020-01-31:
Hi there,
I managed to get my keyplus boards working in wired mode. I want to use one of the boards as the wired receiver and the other board completely wireless.

I can not get the wireless working. I tried everything the last couple hours. The RF settings and the layout should be correct:

```yaml
# Generated on 2020-06-31 at 20:06

rf_settings:
  aes_encryption_key: encryptionkey
  auto_retransmit_count: 15
  data_rate: 2mbps
  pipe0: bcf32b924c
  pipe1: 1f7b477795
  pipe2: '74'
  pipe3: ae
  pipe4: '71'
  pipe5: '00'
  rf_channel: 119
  transmit_power: 0db

```

```yaml
version: 0.3.3

devices:
  receiver:
    id: 0
    layout: receiver
    wired_split: false
    wireless_split: true
    scan_mode: # rf dongle, so don't need a matrix
      mode: pin_gnd
      pins: [
        D3
      ]
  gamepad:
    id: 1
    layout: gamepad
    wired_split: false
    wireless_split: true
    layout_offset: 0
    scan_mode:
      mode: pin_gnd
      pins: [
        D3, D2, D1, D0, A0,
        A1, A2, A3, A4, A5,
        A6, A7, B0, B1, B2,
        B3, C0, C1, C2, C3,
      ]
      # maps how keys are physically wired, to how they appear visually

layouts:
  receiver:
    layers: [
      [ # layer 0 (base)
        [
          x,
        ]
      ],
    ]
  gamepad:
    default_layer: 0
    # This layout shows how layer keys can be chained together. If L1 and
    # L2 are pressed on the base layer, then L3 is activated.
    layers: [
      [ # layer 0 (base)
        [
          esc , 1   , 2   , 3   , 4   ,
          tab , q   , w   , e   , r   ,
          lsft, a   , s   , d   , f   ,
          lctl, x   , g   , c   , spc  ,
        ]
      ],
      # [ # layer 1 (lower)
      #   [
      #     '~' , '!' , '@' , '#' , '$' , '%' , '^' , '&' , '*' , '(' , ')' , ____ ,
      #     del , F1  , F2  , F3  , F4  , F5  , F6  , '_' , '+' , '{' , '}' , '|'  ,
      #     ____, F7  , F8  , F9  , F10 , F11 , F12 , iso~, iso|, home, end , ____ ,
      #     ____, ____, ____, ____, ____, ____, ____, L3  , ____, ____, ____, ____ ,
      #   ]
      # ],
      # [ # layer 2 (raise)
      #   [
      #     '`' ,  1  ,  2  ,  3  ,  4  ,  5  ,  6  ,  7  ,  8  ,  9  ,  0  , ____ ,
      #     del , F1  , F2  , F3  , F4  , F5  , F6  , '-' , '=' , '[' , ']' , '\'  ,
      #     ____, F7  , F8  , F9  , F10 , F11 , F12 , iso#, iso|, pgup, pgdn, ____ ,
      #     ____, ____, ____, ____, L3  , ____, ____, ____, ____, ____, ____, ____ ,
      #   ]
      # ],
      # [ # layer 3 (media 'lower+raise'), press L1+L2 on base layer
      #   [
      #     ____, pscr, mply, volu, mstp, ____, wh_u, btn1, ms_u, btn2, ____, nlck ,
      #     ____, mute, mprv, vold, mnxt, app , wh_d, ms_l, ms_d, ms_r, btn3, slck ,
      #     ____, C-y , CS-z, C-i , ____, ____, ____, ____, ____, ____, ____, caps ,
      #     ____, ____, ____, ____, ____, ____, ____, ____, ____, ____, ____, ____ ,
      #   ]
      # ],
]
```

Can someone maybe look over those and confirm if they look good?
I can also provide all the schematics for the boards I made, but they are basically the same as the keyplus_mini.

Is there any way to debug the wireless connection?

Algram, 2020-02-01:
I also just tried it with a unifying receiver, but it does also not work. At this point I am lost, I don't even know if it is a hardware or a software issue.

Algram, 2020-02-01:
I also just tried it with a unifying receiver, but it does also not work. At this point I am lost, I don't even know if it is a hardware or a software issue.

@ahtn If there is any change you are still around, I would greatly appreciate your help!

angustrau, 2020-02-02:
Have you tried a different RF channel? Try `8`

Algram, 2020-02-02:
@angustrau  I just tried 8 and it also does not work. I should just have to plug in the unifying receiver and then it should work, correct?

angustrau, 2020-02-02:
Pretty sure, yeah

angustrau, 2020-02-02:
I had some trouble with my board where the fab put in a nrf module that was the same size but had a different pinout.

Algram, 2020-02-02:
@angustrau That's something worth checking out. This is the exact model I have:
[image]

How would I check if the pinout is corret?

angustrau, 2020-02-02:
First thing to check is whether the ground pin lines up with the ground plane. For the others you can reference it to the nrf24 pinout from the datasheet.

Algram, 2020-02-02:
Ok let me do that real quick.

Algram, 2020-02-02:
@angustrau I just checked and the pinout is the correct one. Was worth a try though.

Algram, 2020-02-02:
Is there a way to get something like a debug output from the xmega in order to see if the communication with the nrf module is working?

angustrau, 2020-02-03:
Not that I know of. You might need to go digging in the code.

Algram, 2020-02-03:
@angustrau I think I found something: The modules I am using do not actually say "NRF" on them, but rather: `S124R1`

Could that be the root of my problem? I mean I knew I was buying fakes, but this seems like a different chip.

angustrau, 2020-02-03:
It's possible. Maybe plug it into an Arduino and try some test scripts

Algram, 2020-02-05:
@angustrau So I ordered new modules and tried them. Still no luck. Have you designed your own pcb? Can I share my schema and pcb layout for you to check if anything seems odd?

angustrau, 2020-02-06:
Sure I can have a look, although it's been a while since I've done my PCB. You could try this demo script to test your modules https://forum.arduino.cc/index.php?topic=421081.0

Algram, 2020-02-06:
@angustrau  Thanks. I will try the sketches tomorrow. Here is my schematic and pcb:
[image]

[image]

Algram, 2020-02-06:
I think I found something in the code:
https://github.com/ahtn/keyplus/blob/aaeaff3a03ae6e589139829903827be53d9858a0/ports/xmega/src/main.c#L425

Seems like if the usb check pin is active then it always goes into USB mode. I do not have a check pin on my pcb.

Algram, 2020-02-07:
Yep that was it. It is working now. Only the unifying receiver does not work yet.