ahtn/keyplus #107: Comparision with QMK?

danielo515, 2021-12-31:
Hello.
Really interesting project and controllers.
I was wondering how it compares to QMK? I am already very invested on QMK and use a lot of its features, and I am wondering if this firmware is on-par in features.

Regards

LovesTha, 2022-01-01:
The main way it compares poorly to QMK is that it's not been updated in over 2 years.

LovesTha, 2022-01-01:
ZMK is probably the project you're looking for.

danielo515, 2022-01-03:
I ended here because this firmware is what is used with some nice hardware that I found which was used on a wireless dactyl manuform, that's why I got interest

LovesTha, 2022-01-03:
ZMK has plenty of supported promicro compatible controllers supported, it'll probably get you where you want to be :)

Algram, 2022-01-27:
@danielo515 I am successfully running a gamepad and a keyboard setup for over a year now. It is hands down the best keyboard I have every used and the wireless latency is just INSANE.

The only unfortunate thing is, that the project seems to be unmaintained. The author also does not seem to post anything about their keyboards anymore. @ahtn if you are reading, please know that there are at least a few people that absolutely LOVE your take on keyboard software (and hardware in fact).

@danielo515 If you are proficient with coding in python or c, this project does provide something that no other software on the market currently can: Lowest latency 2.4 Ghz keyboard connection that is seamless.
If you are just a consumer of the software I would recommend against using it.