ahtn/keyplus #108: Xmega firmware does not build anymore

Algram, 2022-04-23:
Hi,

I am still a happy user of this project and I was trying to expand to a new device. I can not build the xmega firmware anymore though. This is the error I am getting:

[image]

I am on an Arch VM and followed the instructions at https://github.com/ahtn/keyplus/tree/master/ports/xmega exactly.

@emilytrau  I know you've always been very helpful in the past. Is there any chance you could check of the build still works for you? Thank you so much!

emilytrau, 2022-04-26:
@Algram I tested for a little bit and couldn't get it to build either. If I had to guess it might be that a newer version of GCC has broken the source.

Algram, 2022-08-08:
@emilytrau Damn, thanks for testing thought. I do not really have an idea on how to proceed now. Just hoping my keyboard never dies :rotating_light: